espressif/kicad-libraries #3: Can't open footprint lib

TDA-2030, 2021-07-14:
## Environment

Application: KiCad
Version: 5.1.8-5.1.8, release build
Libraries:
    wxWidgets 3.0.3
    libcurl/7.66.0 OpenSSL/1.1.1d-fips zlib/1.2.11 libidn2/2.2.0 libpsl/0.20.1 (+libidn2/2.2.0) libssh/0.8.7/openssl/zlib nghttp2/1.40.0
Platform: Linux 5.3.18-lp152.78-default x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.3 (wchar_t,STL containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.66.0
    OpenCASCADE Technology: 7.3.0
    Curl: 7.66.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

## Description

I can't open the footprint `Dev_ESP32-S2-Saola-1.kicad_mod` on commit `c8e7b4fa07bac0`. I guess the version is not compatible. Which version of kicad is used for testing?

Following is the error prompt:
```
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 1: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 2: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 3: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 4: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 5: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 6: '=' expected.
11:27:22 AM: file '~/Desktop/kicad-libraries/footprints/Espressif.pretty/Dev_ESP32-S2-Saola-1.kicad_mod', line 7: '=' expected.

Omit similar logs
```

pedrominatel, 2021-07-14:
Hi @TDA-2030.

The library is now for the KiCad v6 (5.99).

My version is: Version: **5.99.0-unknown-395602b12a~131~ubuntu20.04.1, release build**

pedrominatel, 2021-07-14:
@TDA-2030 can you try the legacy branch?

https://github.com/espressif/kicad-libraries/tree/legacy

TDA-2030, 2021-07-15:
I have opened the footprint with KiCad v5.99. It should really be a version compatibility issue. Are the footprints of these two branches the same? 

BTW, I'm adding `ESP32-DevKitc` and `ESP32-S3-DevKitc` to this library as well. If you think it is necessary, I will submit a PR :grin:

pedrominatel, 2021-07-16:
Hi @TDA-2030, thank you for collaborating!

For now, I'm using the `legacy` branch in order to be compatible with V5. Later, we can convert it easily to V6 or let the user do that.
The `feature/new_libraries` will not be updated anymore due to this compatibility issue with V5.

If you already managed to create the libs for the DevKitC on V6 version, you can create the PR on the `feature/new_libraries` and I can convert it back to V5 on the `legacy` branch. Otherwise I will close that branch and keep the `legacy` instead.

TDA-2030, 2021-07-16:
I will create the PR on `legacy` , because KiCad(v5.99) seems to work unsteadily on my computer.

closed 2021-07-16 by pedrominatel