espressif/kicad-libraries #9: Please add ESP-C3-32S

rtreffer, 2021-07-30:
Not sure if this is the right way to ask for it but I was hoping to see footprints for the ESP-C3-32S.

Please add a footprint for that chip (38p with 0.05mm less bottom space than the other WROOMS)

Spritetm, 2021-08-08:
That is an AI-Thinker product as far as I can see, not an Espressif product. You may be better off asking AI-Thinker for a symbol/footprint.

closed 2021-08-08 by rtreffer

MauroMombelli, 2023-09-16:
@Spritetm if I make a merge request, would it be accepted or this repo is only for official layout?

Spritetm, 2023-09-17:
@MauroMombelli I'm no the owner of this repo, but I don't think so... it would be very hard for us to check and maintain footprints for products that are not ours.