espressif/kicad-libraries #24: ESP32-S3-MINI-1 footprint pad 46 not connecting to GND net

atanisoft, 2021-11-18:
All GND pads connect except the non-square pad in the center:
[image]

Checking that specific pad shows that it is explicitly configured to not connect to nets.

pedrominatel, 2021-11-19:
Thank you for your feedback. I'll check it ASAP.

pedrominatel, 2021-11-19:
Hi @atanisoft, KiCad does not support thermal relief on custom shape pads. You can add manually or edit the pad characteristics by double-clicking on the custom pad and setting.

Pad Properties -> Connection to Copper Zone -> Pad Connection -> Solid

[image]

If this works for you, you can close this issue.

Thank you!

atanisoft, 2021-11-19:
@pedrominatel That is rather unfortunate on the KiCad side, but understandable considering the reason why (thermal relief spokes would be challenging to calculate on random polygons). After reading your suggestion I went looking for options to augment the footprint so it can fill automatically like others and one suggestion was to add a second pad (square or circle) that is smaller than the custom pad and stack them. I'll see if I can modify the footprint to have a smaller pad on top of this one and confirm if it works and doesn't generate any unnecessary DRC alerts.

pedrominatel, 2021-11-19:
This smaller pad technique works very well in most of the cases. If you want, you can create a PR for the S3 and we can replicate it for the others.

atanisoft, 2021-11-19:
I'll send a PR in a bit after confirming no DRC related alerts

atanisoft, 2021-11-19:
@pedrominatel in my testing of stacked footprints locally it doesn't seem to connect to the copper fill. I'm thinking your alternative above may be a better option and include it by default in the footprint.

Minimal diff:
```
   (pad 46 smd custom (at -1.65 -1.65) (size 0.6 0.6) (layers F.Cu F.Paste F.Mask)
-    (zone_connect 0)
+    (zone_connect 2)
```

For whatever reason KiCad wants to reorder all of the `fp_line` entries in the footprint which makes little sense since they are the same values just reordered.

pedrominatel, 2021-11-22:
Hi @atanisoft, I can modify that in my next review. I think it's a good idea to add some notes about it on the `README.md` in order to explain this KiCad 5 limitation. Not sure if this happens on KiCad 6.

closed 2021-12-01 by pedrominatel