espressif/kicad-libraries #26: ESP32-C3-WROOM-02U 3D model

BitSalat, 2021-11-23:
Would be great to have a 3D model (step) for the -02U module (without PCB antenna)
as well as we are just implementing it in a commercial design - thank you a lot.

alank-hisky, 2021-11-23:
Found a possible 3rd party solution "ESP-WROOM-02U" STEP file
https://www.3dcontentcentral.com/secure/download-model.aspx?catalogid=171&id=999269
[image]
[image]

pedrominatel, 2022-01-19:
The 3D model will be added on the next revision.

Here is the 3D model: https://www.espressif.com/sites/default/files/3dmodel/ESP32-C3-WROOM-02U_20210906_0.STEP

pedrominatel, 2022-03-04:
Done!