espressif/kicad-libraries #49: ESP32-S3-WROOM2

higaski, 2022-07-01:
Any chance we can expect the WROOM2 versions of the S3 modules any time soon? From what I've seen they have slightly rearranged ground pads in the middle compared to the WROOM1.

pedrominatel, 2022-07-11:
Hi @higaski,

The [ESP32-S3-WROOM-2](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) will be added according to the datasheet landing pattern.

[image]

higaski, 2022-07-11:
Ok, let me rephrase my question.

In case I need a footprint of this part within the next 2-3 weeks, should I better draw it myself or can I count on you guys doing it. :stuck_out_tongue_closed_eyes:

pedrominatel, 2022-07-11:
We can manage that in the next week or soon.

pedrominatel, 2022-07-29:
Done, @higaski. PTAL

PR #51 (2022-07-29): Added ESP32-S3-WROOM-2 footprint and symbol
Added ESP32-S3-WROOM-2 footprint and symbol

Closes #49
closed 2022-07-29 by pedrominatel