espressif/kicad-libraries #56: esp32-mini-1 symbol / footprint

tenderlove, 2022-08-21:
Hi,

I'm working on a project that incorporates the [esp32-mini-1](https://www.espressif.com/sites/default/files/documentation/esp32-mini-1_datasheet_en.pdf), and I was wondering if there is a KiCad symbol / footprint available for that module?  I don't see one in this repository, but I would be happy to make one if there is a similar module I can copy from.

Any help would be greatly appreciated.

Thanks!

pedrominatel, 2022-08-23:
Thanks, @tenderlove. We can provide the new footprint/symbol for this new module.
This module has a different landing pad number from all the other MINI versions. Copy is not possible.

tenderlove, 2022-08-23:
@pedrominatel awesome, thank you!!! 😍

tenderlove, 2022-10-10:
@pedrominatel thanks for fixing this!  I don't see the mini-1 in the repo though.  Am I missing something?  Thanks.

closed 2022-10-28 by pedrominatel