espressif/kicad-libraries #78: Keepout zones over antenna regions

DanielVoogsgerd, 2023-02-22:
In the datasheets for various modules one can find that there is a keepout zone where the antenna is, this makes sense. However, I don't see such a keepout zone in the footprints. Is this for a specific reason I'm not seeing? If it isn't, it might be a good addition to the footprints.

pedrominatel, 2023-02-23:
Hi @DanielVoogsgerd,

I believe this will bring good value to the designers. I'll add the keepout zones for the new fp's and review/bug fix the others.

DanielVoogsgerd, 2023-02-23:
Terrific, if I can be of any assistance feel free to say so. I suspect however that reviewing the keepout zones would be about as much work as drawing them. If it isn't I'm happy to create a PR with the changes.

PR #92 (2023-05-19): Added keepout area for the antenna on each module
This PR adds the keep-out areas for the antenna. 

[image]

Closes #78
closed 2023-05-19 by pedrominatel