espressif/kicad-libraries #81: ESP32-S3-WROOM-1U Symbol

unixb0y, 2023-03-12:
I noticed there is no dedicated symbol for ESP32-S3-WROOM-1U.
Are we supposed to just choose the ESP32-S3-WROOM-1 (without **U**) symbol and change the footprint to ESP32-S3-WROOM-1U? It does seem to work.

Just trying to clarify whether intentional or not.

pedrominatel, 2023-04-18:
Hi @unixb0y,

Both are pin-to-pin compatible, so there is no need to add a new symbol.
We can add a note in the description or change the name to ESP32-S3-WROOM-1/U.

cmcskimming, 2024-02-16:
It would be nice to just have the U variant in the list of options, not having it just adds the extra step of having to edit the symbol and change its footprint to 1U. I'm sure the extra space it takes up would not be an issue for most people when the 1U footprint is already present in the library anyway.

pedrominatel, 2024-02-22:
Hi @cmcskimming,
The extra antenna area is set as keepoout zone. If you need the 1U for some extra space and place the module somewhere that is not on the border of the PCB, you will be not able to use the antenna area.
The most common FP is the U variant, and it's set as default.