espressif/kicad-libraries #97: Linux install

Jef91w, 2023-05-24:
Hi all

I try to install on KiCad : The last version on Linux but it's does not wok any idea please
Regards

Application: KiCad

Version: 6.0.8+dfsg-1~bpo11+1+rpt1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.74.0 OpenSSL/1.1.1n zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3

Platform: Linux 6.1.21-v8+ aarch64, 64 bit, Little endian, wxGTK, lightdm-xsession, x11

Build Info:
	Date: Oct  7 2022 14:52:26
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.74.0
	ngspice: 34
	Compiler: GCC 10.2.1 with C++ ABI 1014

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

pedrominatel, 2023-05-24:
Hi @Jef91w,

Have you tried this branch?

https://github.com/espressif/kicad-libraries/tree/legacy_kicad6

Jef91w, 2023-05-25:
Hi 
I'm stupid many thanks

;-)

closed 2023-05-26 by pedrominatel