espressif/kicad-libraries #107: Add kicad repository manifest for the plugin manager

dzid26, 2023-06-17:
Add `repository.json`, etc. 
This will allow the updates to be handled by kicad plugin manager. 

Here is an example:
https://github.com/Bouni/kicad-jlcpcb-tools#kicad-pcm

pedrominatel, 2023-06-18:
Hi @dzid26, this task is already in progress.

hayschan, 2023-12-27:
Any update on this? This will allow easier update instead of manually coming to GitHub and download the latest file.

JarrettR, 2024-05-23:
The next blocker here is that the KiCad PCM doesn't allow STEP files to have an uppercase extension.

It's `.step` or nothing. I have requested that requirement be relaxed by the KiCad team, and the request has been denied.

Steffen-W, 2025-02-13:
@JarrettR, why is it not possible to change the file extension to .step? Isn't the work really not that big?

JarrettR, 2025-02-16:
> [@JarrettR](https://github.com/JarrettR), why is it not possible to change the file extension to .step? Isn't the work really not that big?

Here's the MR where I did that, but it was requested to revert that part of it:
https://github.com/espressif/kicad-libraries/pull/131#issuecomment-1793523041

It's worth a discussion with @pedrominatel if a new MR to fix that would be accepted

pedrominatel, 2026-11-12:
Related to #209

closed 2026-11-12 by pedrominatel