espressif/kicad-libraries #112: Archive does not contain a valid metadata.json file

muscatelle, 2023-06-30:
I've just tried to install this version on kicad 7.05 as listed below and get an error message:
"Archive does not contain a valid metadata.json file"

Tony

Application: KiCad x86_64 on x86_64

Version: 7.0.5-7.0.5~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.14

Platform: Ubuntu 22.04.2 LTS, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: May 31 2023 23:56:22
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

UnexpectedMaker, 2023-07-02:
Getting the same - KiCAD 7.0.5 on macOS.

pedrominatel, 2023-07-03:
Hi,

Could you please confirm (unzipping) if the zip file contains the `metadata.json`? Can you send me the link that you are using to download the file?

UnexpectedMaker, 2023-07-03:
> Hi,
> 
> Could you please confirm (unzipping) if the zip file contains the `metadata.json`? Can you send me the link that you are using to download the file?

It does have the `metadata.json` file in the archive, but KiCAD 7.05 (at least on the Mac) doesn't like it.

I'm using the releases link form the releases page...
https://github.com/espressif/kicad-libraries/releases/download/2.0.0/espressif-kicad-addon.zip

Cheers :)

pedrominatel, 2023-07-03:
I've tested right now on macOS Monterey (12.5) Apple M1 chip and it worked on KiCad 7.0.5.
I'll ask someone else with macOS to test it and see if we can identify the issue.

Anyway, I'll release a new version (2.0.1) with some fixes later today. Maybe you can try with the new zip file.

Hi-Rob, 2023-07-03:
Hi I got the same error on win7
there were a metadata.json file in the .zip

pedrominatel, 2023-07-03:
Could you please test this pre-release 2.0.1 version?

https://github.com/espressif/kicad-libraries/releases/download/2.0.1/espressif-kicad-addon.zip

UnexpectedMaker, 2023-07-03:
> Could you please test this pre-release 2.0.1 version?
> 
> https://github.com/espressif/kicad-libraries/releases/download/2.0.1/espressif-kicad-addon.zip

Doesn't work...
[image]

2023 16" MacBookPro with M2 MAX running macOS 13.4.1 and KiCAD 7.0.5

Hi-Rob, 2023-07-03:
The addon

> Could you please test this pre-release 2.0.1 version?
> 
> https://github.com/espressif/kicad-libraries/releases/download/2.0.1/espressif-kicad-addon.zip

This worked to me

muscatelle, 2023-07-10:
Now works for me also using KiCad x86_64 on x86_64 on 7.0.5-7.0.5~ubuntu22.04.1, release build

Thanks

murarduino, 2023-07-10:
[image]

zip file:
[image]

serant, 2023-08-10:
I think I have a way to reproduce this issue.

I had the same issue when downloading v2.0.1 using **Safari** on my MacBook which automatically unzips the folder after downloading. I re-zipped the folder and ran into the issue described in this thread when importing the library.

I then disabled "Open "safe" files after downloading" in Safari's preferences to prevent the zip files from automatically being extracted because I had a feeling that me re-zipping the files was causing an issue. This time v2.0.1 works flawlessly!

UnexpectedMaker, 2023-08-10:
@serant Well done! That was it!

pedrominatel, 2023-08-16:
This explains the issue. The GitHub CI generates the ZIP file and contains the file hash inside the ``metadata.json`` file to make sure the file is not corrupted or something else. If you unzip and zip again, the hash will differ from the one inside the metadata.

This issue is not related to the library itself. Thank you @serant for describing it.
I'll close this issue in 24 hours if there is no more activity here.

closed 2023-08-17 by pedrominatel