espressif/kicad-libraries #123: Which footprint for ESP32-S3-PICO

lclaude, 2023-10-09:
Hello,
I'm not sure which footprint to use with the ESP32-S3-PICO.
In the datasheet the package is LGA56 (7x7mm), I can't find an equivalent in your footprints.
Thanks by advance.

pedrominatel, 2023-10-10:
Hi @lclaude,

You can use the **QFN-56 1EP 7x7mm P0.4mm EP4x4mm** from the official KiCad library.

**QFN-56-1EP_7x7mm_P0.4mm_EP4x4mm**

[image]

Equivalent to the LGA-56

[image]

lclaude, 2023-10-10:
OK, I'll test with this one. Thank you.

PR #181 (2025-02-04): added esp32-c5-devkitc symbol and footprint to library

## Description

Added ESP32-C5-DevKitC Kicad symbol and footprint. 

## Related
closed 2023-10-10 by lclaude