espressif/kicad-libraries #127: GPIO11 in ESP32-C6-WROOM-1

david-varela, 2023-10-22:
The ESP32-C6-WROOM-1 symbol has some pins as invisible. One of those is the pin **GPIO11**, pin number 12, of type bidirectional, which **is placed in the same position as the grounds**, pins number 1, 28 and 29. See details in the table below:

| Pin name | Pin number | Electrical type | Visible | X position | Y position |
| --- | --- | --- | --- | --- | --- |
| GND | 1 | Power input | Yes | 0 | 33.02 |
| GPIO11 | 12 | Bidirectional | No | 0 | 33.02 |
| GND | 28 | Passive | No | 0 | 33.02 |
| GND | 29 | Passive | No | 0 | 33.02 |

I believe, looking at the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c6-wroom-1_wroom-1u_datasheet_en.pdf), that the pin number 12 should be visible and in a different location, am I right? Right now, when I connect pin number 1 to ground of my battery, the rules checker shows a warning for connecting pin  number12 to ground, because pin number 12 is invisible and shares the location with pins 1, 28 and 29.

closed 2023-10-22 by pedrominatel

pedrominatel, 2023-10-22:
Thank you @david-varela for reporting this issue.
Fixed and new version released! Please update to the version 2.0.4.

david-varela, 2023-10-22:
Thank you @pedrominatel for the fix and the new release!

PR #128 (2023-10-22): ESP32-C6-WROOM-1 pin12 fix
Fix and close #127