espressif/kicad-libraries #132: Archive does not contain a valid metadata.json file

meakashrao, 2023-11-14:
Hello, I tried to install from Main Release or Branch Release and metadata error keeps popping up. 

Please advise what to do.
[image]

Regards,
Akash Rao

pedrominatel, 2023-11-14:
Hi,
Can you check this issue #105 if it is related to yours?

meakashrao, 2023-11-14:
No, I don't think so, I went ahead and downloaded 7.0.9 Kicad, could you please help me with the latest release branch so that I can try that out

meakashrao, 2023-11-14:
https://github.com/espressif/kicad-libraries/releases/tag/2.0.4

We'll sorry this did work for me.

Thanks for the help

closed 2023-11-14 by meakashrao