espressif/kicad-libraries #142: KiCAD error. Archive does not contain a valid metada.json file

fjrg76-com, 2024-01-02:
Hi, I've just downloaded the project and while attempting to open it with the KiCAD7's PCM manager I got the above error.

Did I miss something? Any hint?

pedrominatel, 2024-01-03:
Hi,

**If you are using macOS:** _Make sure to download the correct zip file and do not extract the files. If you are using macOS and Safari, ensure to that the automatic unzip feature (Open safe files after downloading) is disabled._

zerovijay, 2024-01-04:
use this [link](https://github.com/espressif/kicad-libraries/releases/download/2.0.6/espressif-kicad-addon.zip) 
download the zip file

pedrominatel, 2024-01-08:
Any update, @fjrg76-com?

grantgumina, 2024-01-14:
Ran into the same issue and disabling Safari's "auto open" feature worked great.

Navarro-Franco, 2024-01-22:
I found a partial solution. 
1) Clone the repo.
`git clone https://github.com/espressif/kicad-libraries`
2) Go to the repo and run the script with python
`cd kicad-libraries/ &&  python package.py`
you should get an output like this:
```
This script helps to generate a new kicad addon release.
It generates the release zip file and the metadata.json
Enter the new addon version (required format: major[.minor[.patch]]): 

```
3) Enter a version like 2.0.7
`Enter the new addon version (required format: major[.minor[.patch]]): 2.0.7`
( note: it must be higher than the version you download, in my case, the latest version is the 2.0.6)
The script will generate a folder called " build " and a zip file with a valid metada.json

pedrominatel, 2024-01-23:
Use this link to download the latest version.

### [Download here the latest library!](https://github.com/espressif/kicad-libraries/releases/latest/download/espressif-kicad-addon.zip)

Do not unzip the file and zip again and disable the auto-open/opan-safe function on Safari.

dacut, 2024-01-31:
I encountered this myself (on Chrome). In my case, my mistake was going to the Tags and selecting the zip link under the 2.0.6 tag, because GitHub hides the releases off to the side of the UI. This issue helped me find the right link to use.

closed 2024-05-29 by pedrominatel