espressif/kicad-libraries #178: ESP32-P4 pin misalignment

PickaxeHit, 2025-01-06:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

none

### Actual behavior (suspected bug)

[image]

### Steps to reproduce the behavior

Install the package and open ESP32-P4 symbol and it's here.

### Project release version

v3.0.1 release

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

arch+kicad8.0.7

### Shell

ZSH

### Additional context

The positions of GPIO5 and GPIO6 are swapped (although they still correspond to the pin numbers, they do not meet the design specifications)

pedrominatel, 2025-01-07:
Thank you very much for your issue report.
This is indeed a bug and it will be fixed as soon as possible.

pedrominatel, 2025-01-07:
> Thank you very much for your issue report. This is indeed a bug and it will be fixed as soon as possible.

This bug has no impact on the footprint, the pin number, and the GPIO matches. The issue is only the order on the symbol.

closed 2025-01-09 by pedrominatel