espressif/kicad-libraries #187: ESP32-C3-MINI-1U not included in Espressif.kicad_sym

DarranShepherd, 2025-05-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected to be able to select ESP32-C3-MINI-1U when placing a symbol on a schematic.

### Actual behavior (suspected bug)

The ESP32-C3-MINI-1 module is available, but the ESP32-C3-MINI-1U is not.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Install the library in Kicad 8.0.9
Click Place Symbol on the schematic editor
Search for ESP32-C3-MINI
ESP32-C3-MINI-1 is available, but ESP32-C3-MINI-1U is not

[image]

### Project release version

3.0.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

I can see the footprint and 3d model files, but noticed that the ESP32-C3-MINI-1U is not included in the symbols file. I manually edited the file to copy the ESP32-C3-MINI-1 symbol and replaced ESP32-C3-MINI-1 with ESP32-C3-MINI-1U where necessary and this does now show up when choosing a symbol, but the pads are inset from the edge of the package and I'm very much out of my depth at this point...

[image]

Apologies if this is a beginner user mistake.

DarranShepherd, 2025-05-12:
I can now see that the inset pads on one edge matches the datasheet, so perhaps my manual edit of the Espressif.kicad_sym file is sufficient?

kozmotronik, 2025-07-25:
+1. I need to use the `ESP32-C3-MINI-1U` in a PCB. The only module missing in the `symbols` library. Although the pins are identical for both, using the `ESP32-C3-MINI-1` in the schematic and the `ESP32-C3-MINI-1U` in the PCB may lead to confusions. So please consider adding the `symbol` as well.

Kind regards

closed 2026-11-13 by pedrominatel